adafruit/Adafruit_Python_BluefruitLE #5: hcitool lescan shows BLE device but python list_uarts.py does not

jpummill, 2015-09-30:
Just starting to learn about working with Bluetooth BLE devices and I am having an issue.

hcitool lescan shows BLE device but python list_uarts.py does not (see below):

hciconfig example:
--------
pi@rasp01 ~/bluefruitle/Adafruit_Python_BluefruitLE/examples $ hciconfig
hci0:   Type: BR/EDR  Bus: USB
        BD Address: 5C:F3:70:6C:3A:DC  ACL MTU: 1021:8  SCO MTU: 64:1
        UP RUNNING
        RX bytes:2591 acl:0 sco:0 events:164 errors:0
        TX bytes:1522 acl:0 sco:0 commands:128 errors:0
hcitool lescan example:
---------
pi@rasp01 ~/bluefruitle/Adafruit_Python_BluefruitLE/examples $ sudo hcitool lescan
LE Scan ...
74:DA:EA:AC:96:68 (unknown)
74:DA:EA:AC:96:68 HMSensor
python list_uarts.py example:
---------
pi@rasp01 ~/bluefruitle/Adafruit_Python_BluefruitLE/examples $ sudo python list_uarts.py
Using adapter: BlueZ 5.33
Searching for UART devices...
Press Ctrl-C to quit (will take ~30 seconds on OSX).
(nothing shows up here after waiting for 30 to 40 seconds so I use Ctrl-C to end process)

P.S.  I am using the following BLE device from Amazon/Diymall.
http://www.amazon.com/Diymall%C2%AE-Bluetooth-Near-field-Positioning-Acquisition/dp/B00O2UDZU6

microbuilder, 2015-09-30:
The example is doing what it says ... looking for 'UART' devices, which means BLE devices advertising with a specific UART Service UUID (see https://github.com/adafruit/Adafruit_Python_BluefruitLE/blob/master/Adafruit_BluefruitLE/services/uart.py).  Since you aren't using a Bluefruit LE module with this code, it isn't finding any matches, and you'll need to update the search parameters to match your own HW.

closed 2015-09-30 by microbuilder

jpummill, 2015-09-30:
Ok, thank you.