adafruit/Adafruit_Python_BluefruitLE #10: TypeError exception on Device.disconnect() for Sphero BB-8 on OS X

wolfteeth, 2016-01-20:
Build off from one of the example scripts to connect with a Sphero BB-8 toy, list its advertised BLE services, and disconnect. On calling disconnect(), I see this error:

    Traceback (most recent call last):
      File "/Users/kris/Development/Projects/pysphero/src/test.py", line 58, in <module>
        provider.run_mainloop_with(main)
      File "build/bdist.macosx-10.11-intel/egg/Adafruit_BluefruitLE/corebluetooth/provider.py", line 265, in run_mainloop_with
      File "/Library/Python/2.7/site-packages/PyObjCTools/AppHelper.py", line 182, in runConsoleEventLoop
        if not runLoop.runMode_beforeDate_(mode, nextfire):
      File "/Library/Python/2.7/site-packages/PyObjCTools/AppHelper.py", line 41, in call_
        func(*args, **kwargs)
      File "build/bdist.macosx-10.11-intel/egg/Adafruit_BluefruitLE/corebluetooth/provider.py", line 274, in _user_thread_main
      File "/Users/kris/Development/Projects/pysphero/src/test.py", line 53, in main
        bb8.disconnect()
      File "build/bdist.macosx-10.11-intel/egg/Adafruit_BluefruitLE/corebluetooth/device.py", line 73, in disconnect
      File "build/bdist.macosx-10.11-intel/egg/Adafruit_BluefruitLE/corebluetooth/gatt.py", line 59, in list_characteristics
      File "build/bdist.macosx-10.11-intel/egg/Adafruit_BluefruitLE/corebluetooth/metadata.py", line 63, in get_all
    TypeError: 'NoneType' object is not iterable

My test scripts are here: https://gist.github.com/wolfteeth/9baf0133c582fc08e518

I'm not sure if I'm doing something wrong, or if there's something unusual about the device that the library wasn't ready for.

enborra, 2016-01-31:
Was investigating similar task (and resulting error) and just found that this error gets thrown when DeviceInformation.discover hasn't been run before disconnect is called. The disconnect method seems to consider it an exception throw if it doesn't see pre-existing metadata...interrogating the device service before calling disconnect cleared up the error for me:

In your gist, this may fix the exception..

`
    bb8.connect()
    print 'Connected: ' + str(bb8.is_connected)
    print 'Advertised services: ' + str(bb8.advertised)

    print('Discovering services...')
    DeviceInformation.discover(bb8)

    # Once service discovery is complete create an instance of the service
    # and start interacting with it.
    dis = DeviceInformation(device)

  bb8.disconnect()
`

/may the force be with you/

bpolonia, 2017-03-21:
Is there a way to reconnect after disconnecting? I'm running into an issue where I connect to a device, start discovery, create an instance of the service, disconnect, but then if I call connect on it again it fails to connect.

closed 2020-11-02 by dhalbert