adafruit/Adafruit_Python_BluefruitLE #11: Linux Performance Issues due to Repeated Polling of ObjectManager GetManagedObjects()

mattthebaker, 2016-03-26:
I'm working on an application that requires repeated connection, discovery, some minor activity and disconnection.

Every 4-5 seconds or so, the stack locks up for 15 or so more seconds.  Host in question is a Raspberry Pi 1.

My debugging traced this back to GetManagedObjects() being slow to respond and sometimes locking up.

From what I can tell reading the [DBus docs on ObjectManager](https://dbus.freedesktop.org/doc/dbus-specification.html#standard-interfaces-objectmanager), they never intended for GetManagedObjects() to be called repeatedly -- rather, it is to enable a quick initial sync of the entire state of the endpoint, and then changes are to be communicated by InterfacesAdded and InterfacesRemoved events.  

In device.discover() for example, the code calls GetManagedObjects() every second until the endpoints it is looking for appear.  In one sample of my connection through disconnection cycle, there were 13 calls, 5 of which during discover().  As other device enter the system and end up in the device tree, the performance would degrade even faster.

The least invasive change for a big benefit would be to maintain the complete dict returned by GetManagedObjects() inside of the provider, updating it on Interface* events.  For a full and optimized BLE implementation maybe it should be implemented at the Adapter or Device level using the AddWatch mechanism mentioned in the docs.

You guys have implemented the most complete Python BLE library, and out of everything out there it is really the only one to implement it properly for Linux/Bluez i.e. using DBus to bluetoothd and not helper c wrappers.  This could (should) be morphed into the standard Python BLE implementation.

I may take a crack at a proof of concept implementation in the provider.

Cheers

neomanic, 2017-07-06:
How have you got on with this, @mattthebaker? I'm having the same issues, I think, as it's glacially slow to search for the services and characteristics.

I agree with your assessment that this is the most complete Python BLE library. All the others are useless since they haven't kept up with BlueZ's advancement and don't used the recommended D-Bus APIs. I did my original development on macOS, where it is even more reliable and then switched it over to Linux/Ubuntu for deployment with only a few relatively simple (in hindsight only, mind you!) issues to fix.

mattthebaker, 2017-07-07:
@neomanic 

I refactored my local copy to avoid the repeated calls to GetManagedObjects(), which improved my first performance problem.

After further testing, I've concluded that D-Bus is worthless for my use case.  Basically, the arbitration overhead of D-Bus makes it woefully inadequate for handling a high volume of messages.  

Normally, if you're using a BT SIG standard device, the driver will have been implemented in C as part of BlueZ.  So for example, a BT mouse would not require every data packet to be pushed through D-Bus.  Only the pairing or initial setup would utilize D-Bus.  For that use case it is highly desirable, as you could pair your mouse using the desktop gui and have it work permanently system wide.

I was implementing a custom protocol, so every notification was sent through the D-Bus stack.  The overhead of D-Bus is extreme and performance dismal.  On a Raspberry Pi 1, I could stream data from my device at full speed using bluepy (uses a C slave helper exe linked against bluez) with cpu use at 1% or less.  The same using D-Bus dropped half of my data packets, and the cpu use was 100%.

If I were starting again now, pygattlib (python wrapper for a c++ gatt library) seems promising.  However, they are switching to D-Bus for all future versions as the BTLE backend, so we are back to square one.  

We are working on an official python API for our products, and part of that will be officially putting our support behind a python BTLE implementation (which one TBD).

neomanic, 2017-07-18:
Hey @mattthebaker, sorry for the slow reply, was on leave for the last week. Thanks for sharing your issues, bit sad to hear of the lousy performance.

I tried using pygattlib before I came over to using BluefruitLE (and because I did the original tests on another vendor's nRF51 dev board before switching to the nRF52 Bluefruit), and it failed pretty hard to the point where I gave up. AFAICT pygattlib, and indeed gattlib, and the previous standard bluez tools of hcitool and gatttool have been deprecated.

That said, my application isn't particularly intensive: it's some basic IO operations. Once it's up and running, it goes okay, just takes forever to initialise some times. Any chance you can share your changes to this library for me to leverage off?

(ps having a browse of your e-wall repos... looks very interesting)

mattthebaker, 2017-07-18:
Hi John,

We've had good results with pygattlib.

I would say that DBus chokes on 100Hz BLE notifications on a Pi3 which is
pretty sad given a quad core 1Ghz.

I need to squash the commits and remove some proprietary stuff but I will
send you the final versions of the files that were changed via email.

Cheers.

ukBaz, 2017-07-27:
Hello,

I noticed with the latest release of BlueZ (5.46) they have introduced [AcquireNotify()](https://git.kernel.org/pub/scm/bluetooth/bluez.git/tree/doc/gatt-api.txt#n117) and [AcquireWrite()](https://git.kernel.org/pub/scm/bluetooth/bluez.git/tree/doc/gatt-api.txt#n94) to the DBus API which seems to allow the option of data being sent via sockets rather than through DBus in high volume situations.

Would this would allow for faster throughput of data?

Cheers.

closed 2020-11-02 by dhalbert