adafruit/Adafruit_Python_BluefruitLE #15: Raspberry Pi Connection Error

rselwyn, 2016-07-27:
- Platform/operating system (i.e. Raspberry Pi with Raspbian operating system,
  Windows 32-bit, Windows 64-bit, Mac OSX 64-bit, etc.):  **Raspbian running on A Raspberry Pi 3**

- Python version (run `python -version` or `python3 -version`):  **2.7.9**

- Error message you are receiving, including any Python exception traces:  
```
Traceback (most recent call last):
  File "device_info.py", line 89, in <module>
    ble.run_mainloop_with(main)
  File "build/bdist.linux-armv7l/egg/Adafruit_BluefruitLE/bluez_dbus/provider.py", line 118, in _user_thread_main
  File "device_info.py", line 66, in main
    dis = DeviceInformation(device)
  File "build/bdist.linux-armv7l/egg/Adafruit_BluefruitLE/services/device_information.py", line 58, in __init__
AttributeError: 'NoneType' object has no attribute 'find_characteristic'
```

followed by the inability to connect to the device again.

# Reproduce the Issue.

1. Raspberry Pi 3 running Raspbian
2. nRF51822 powered circuit board
3. Using a board that has never connected to the Raspberry Pi, go to the examples folder and run `python device_info.py` (will crash with above stack trace). 
4.  Attempt to connect to it again using `python low_level.py`.  It won't connect.

Note: This will run on a mac, but not on a Raspberry Pi 3.

fraguada, 2016-09-07:
I have the same problem, same platform, same behavior and I'm desperately seeking a solution to connect to a specific bt mac address (I don't need device detection) using uart.  The list_uatrs.py script find the device, the uart_service.py script gives the error.  Running bluez-5.23 because installing other versions seem to loose built in bluetooth on the pi.

ukBaz, 2017-01-04:
@fraguada, Did you solve your issue with installing newer versions of BlueZ on a RPi3? 

I find I have to patch BlueZ after dong the `configure` and before doing the `make && sudo make install` step.
The patch is at:
https://gist.github.com/pelwell/c8230c48ea24698527cd

The basic instructions I follow are:
```
$ cd bluez-5.43/
$ wget https://gist.github.com/pelwell/c8230c48ea24698527cd/archive/3b07a1eb296862da889609a84f8e10b299b7442d.zip
$ unzip 3b07a1eb296862da889609a84f8e10b299b7442d.zip
$ git apply  -v c8230c48ea24698527cd-3b07a1eb296862da889609a84f8e10b299b7442d/*
$ make && sudo make install
$ sudo reboot
```

closed 2020-11-02 by dhalbert