adafruit/Adafruit_Python_BluefruitLE #16: Python 3 installation failure

watsonix, 2016-09-14:
Summary: Build error on Python 3 because of raise statement syntax

- Platform/operating system (i.e. Raspberry Pi with Raspbian operating system,
  Windows 32-bit, Windows 64-bit, Mac OSX 64-bit, etc.):  

OSX 10.10

- Python version (run `python -version` or `python3 -version`):  3.5

- Error message you are receiving, including any Python exception traces:  

byte-compiling build/bdist.macosx-10.5-x86_64/egg/Adafruit_BluefruitLE/bluez_dbus/provider.py to provider.cpython-34.pyc
  File "build/bdist.macosx-10.5-x86_64/egg/Adafruit_BluefruitLE/bluez_dbus/provider.py", line 103
    raise self._exception[1], None, self._exception[2]
                            ^
SyntaxError: invalid syntax

byte-compiling build/bdist.macosx-10.5-x86_64/egg/Adafruit_BluefruitLE/config.py to config.cpython-34.pyc
byte-compiling build/bdist.macosx-10.5-x86_64/egg/Adafruit_BluefruitLE/corebluetooth/__init__.py to __init__.cpython-34.pyc
byte-compiling build/bdist.macosx-10.5-x86_64/egg/Adafruit_BluefruitLE/corebluetooth/adapter.py to adapter.cpython-34.pyc
byte-compiling build/bdist.macosx-10.5-x86_64/egg/Adafruit_BluefruitLE/corebluetooth/device.py to device.cpython-34.pyc
byte-compiling build/bdist.macosx-10.5-x86_64/egg/Adafruit_BluefruitLE/corebluetooth/gatt.py to gatt.cpython-34.pyc
byte-compiling build/bdist.macosx-10.5-x86_64/egg/Adafruit_BluefruitLE/corebluetooth/metadata.py to metadata.cpython-34.pyc
byte-compiling build/bdist.macosx-10.5-x86_64/egg/Adafruit_BluefruitLE/corebluetooth/objc_helpers.py to objc_helpers.cpython-34.pyc
byte-compiling build/bdist.macosx-10.5-x86_64/egg/Adafruit_BluefruitLE/corebluetooth/provider.py to provider.cpython-34.pyc
  File "build/bdist.macosx-10.5-x86_64/egg/Adafruit_BluefruitLE/corebluetooth/provider.py", line 290
    raise exec_info[1], None, exec_info[2]
                      ^
SyntaxError: invalid syntax

byte-compiling build/bdist.macosx-10.5-x86_64/egg/Adafruit_BluefruitLE/interfaces/__init__.py to __init__.cpython-34.pyc

- List the steps to reproduce the problem below (if possible attach code or commands
  to run): 

%sudo python setup.py install

watsonix, 2016-09-14:
see https://www.python.org/dev/peps/pep-3109/ for how to correctly do this in 3. Line 290 in Adafruit_BluefruitLE/corebluetooth/provider.py should become:

        e = self.exec_info[1](None)
        e.__traceback__ = self.exec_info[2]
        raise e

snimmagadda1, 2016-10-03:
I am also getting the same result. I changed line 290 to above but am still getting the same result..

pgebhard, 2016-10-26:
How do you have the same result. Did you not reinstall what you just modified? 
I also found I had to make a similar change in bluez_dbus/provider.py on line 103.

pgebhard, 2016-10-26:
I also had to add some dot characters in front of some of the module names in __init__.py and services/__init__.py

watsonix, 2016-10-26:
FWIW, I never fully got this running. Ran out of steam on it and went with Bluetooth via browser (Chrome) instead: https://developer.chrome.com/apps/app_bluetooth

Here are my totally unfinished notes which may or may not be much help:

> 
> 2to3 -w . 
> 
> 	Print statements will now have extra parens (but whatever)
> 
> pip install pyobjc
> 
> Adafruit_BluefruitLE/corebluetooth/adapter.py", line 39: add .encode('utf-8')

snimmagadda1, 2016-11-07:
Thanks for your notes I fixed the packages, though am not sure how to fix the TypeError: a bytes-like object is required, not 'str' generated by line 39 in Adafruit_BluefruitLE/corebluetooth/adapter.py. Where would I add line 39: add .encode('utf-8')?

bpolonia, 2017-01-04:
@snimmagadda1 Did you ever get anywhere with this?
@pgebhard You got this working with python3?

My installation works but when running code using python3 I'm getting errors. However I don't get the errors when I run with python2

I've followed your notes. All the way through @watsonix note about adding `.encode('utf-8')` on line 39 of Adafruit_BluefruitLE/corebluetooth/adapter.py.

`[('IOBluetoothPreferenceGetControllerPowerState'.encode('utf-8'), 'oI'.encode('utf-8')), ('IOBluetoothPreferenceSetControllerPowerState'.encode('utf-8'), 'vI'.encode('utf-8'))]`

I'm getting a `TypeError: functionInfo name not a string`

The full trace is:

```
Traceback (most recent call last):
  File "main.py", line 1, in <module>
    from controllers.ble_controller import BleController
  File "/Users/brianpolonia/QA/controllers/ble_controller.py", line 12, in <module>
    class BleController(ManufacturerDataConstants, ServicesController, Mesh):
  File "/Users/brianpolonia/QA/controllers/ble_controller.py", line 15, in BleController
    ble = Adafruit_BluefruitLE.get_provider()
  File "/usr/local/lib/python3.5/site-packages/Adafruit_BluefruitLE-0.9.0-py3.5.egg/Adafruit_BluefruitLE/platform.py", line 42, in get_provider
    #    1.0.2 - added more Windows support
  File "<frozen importlib._bootstrap>", line 969, in _find_and_load
  File "<frozen importlib._bootstrap>", line 958, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 664, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 634, in _load_backward_compatible
  File "/usr/local/lib/python3.5/site-packages/Adafruit_BluefruitLE-0.9.0-py3.5.egg/Adafruit_BluefruitLE/corebluetooth/provider.py", line 333, in <module>
  File "<frozen importlib._bootstrap>", line 969, in _find_and_load
  File "<frozen importlib._bootstrap>", line 958, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 664, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 634, in _load_backward_compatible
  File "/usr/local/lib/python3.5/site-packages/Adafruit_BluefruitLE-0.9.0-py3.5.egg/Adafruit_BluefruitLE/corebluetooth/adapter.py", line 39, in <module>
TypeError: functionInfo name not a string
```

I'm on Mac OS Sierra 10.12.2

Abhipray, 2017-01-26:
From [https://pythonhosted.org/pyobjc/metadata/manual.html#objc.loadBundleFunctions] ,

> functionInfo is a sequence of function descriptions. Every description is a tuple of two or four elements: the function name (a string) and signature (a byte string) and optionally a value for the “__doc__” attribute and a metadata dictionary.

So change it to:
`[(u'IOBluetoothPreferenceGetControllerPowerState', b'oI'),(u'IOBluetoothPreferenceSetControllerPowerState', b'vI')]`

closed 2020-11-02 by dhalbert