adafruit/Adafruit_Python_BluefruitLE #17: Switch between BLE adapters

bedenko, 2016-11-09:
in *provider.py* function *get_default_adapter* returns first adapter by default.

The problem is if there are multiple adapters, or in my case USB-BLE key with Bluetooth **and** BLE functionality.
In this case, first adapter (Bluetooth) is selected by default, and there is no option to switch to BLE.

tabchas, 2016-11-11:
@bedenko any idea for a quick solution to this?

bedenko, 2016-11-12:
Well, at least for linux part, [this part](https://github.com/adafruit/Adafruit_Python_BluefruitLE/blob/15091513d73c0253e843632a3487aa741ccc3146/Adafruit_BluefruitLE/bluez_dbus/provider.py#L162-L169) is hard-coded, so "Adapter1" and "Device1" are static.

Quick fix: change the value **Adapter1** to **Adapter2**
Permanent fix: change the code, to be able to switch between adapters.

P.S.: [this line](https://github.com/adafruit/Adafruit_Python_BluefruitLE/blob/15091513d73c0253e843632a3487aa741ccc3146/Adafruit_BluefruitLE/bluez_dbus/provider.py#L179) also returns following error: 

```python
AttributeError: 'NoneType' object has no attribute 'GetManagedObjects'
```

neomanic, 2017-07-06:
You can use `adapters = ble.list_adapters()`, then choose the adapter you want from the list using indexing `adapter = adapters[-1]`.

But on a related issue, I wasn't sure that the adapters would always be enumerated in the right order, so I added adapter selection using the MAC of the Bluetooth dongle, as follows.

Add the following patch to the library, which gives you easy access to the BLE adapter address:
```python
diff --git a/Adafruit_BluefruitLE/bluez_dbus/adapter.py b/Adafruit_BluefruitLE/bluez_dbus/adapter.py
index 0d262d2..811c0bc 100644
--- a/Adafruit_BluefruitLE/bluez_dbus/adapter.py
+++ b/Adafruit_BluefruitLE/bluez_dbus/adapter.py
@@ -62,7 +62,13 @@ class BluezAdapter(Adapter):
     def name(self):
         """Return the name of this BLE network adapter."""
         return self._props.Get(_INTERFACE, 'Name')
-
+        
+    @property
+    def address(self):
+        """Return the address of this BLE network adapter."""
+        return self._props.Get(_INTERFACE, 'Address')
+        
+        
     def start_scan(self, timeout_sec=TIMEOUT_SEC):
         """Start scanning for BLE devices with this adapter."""
         self._scan_started.clear()
```

Then in my initialisation code, I have:

```python
    adapters = ble.list_adapters()
    adapter = None
    if len(adapters) == 1:
        print("Single Bluetooth adapter found (no MAC filtering).")
        adapter = adapters[0]
    if len(adapters) > 1:
        filtered_adapters = [a for a in adapters if a.address == BLUETOOTH_ADAPTER_MAC]
        if len(filtered_adapters) > 0:
            print("Multiple Bluetooth adapters found: filtered by MAC.")
            adapter = filtered_adapters[0]
        else:
            print("Multiple Bluetooth adapters found: MAC filtering failed, using last enumerated adapter.")
            adapter = adapters[-1]
    elif len(adapters) == 0:
        print("No Bluetooth adapter found.")
        return

```

closed 2020-11-02 by dhalbert