adafruit/Adafruit_Python_BluefruitLE #19: uart_service.py throws an error on Raspberry Pi 3

benlhy, 2016-12-18:
- Raspberry Pi 3

- Python 2.7

- Error message you are receiving, including any Python exception traces:  

File "uart_service.py", line 85, in <module>
  ble.run_mainloop_with(main)
File "build/bdist.linux-armz7l/egg/Adafruit_BluefruitLE/bluez_dbus/provider.py", line 118, in _user_tread_main
File "uart_service.py", line 59, in main
  uart = UART(device)
File "build/bdist.linux-arm7l/egg/Adafruit_BluefruitLE/services/uart.py", line 48, in __init__
AttributeError: 'NoneType' object has no attribute 'find_characteristic'

- List the steps to reproduce the problem below (if possible attach code or commands
  to run): 

Adafruit Bluefruit 32u4 Feather running bleuart_datamode
Run `sudo python uart_service` from the examples folder.
Upon "Discovering Services", this error results.

warren-r, 2017-02-23:
I am also having same problem

cturner48, 2017-04-11:
I was just having a similar problem. I am not sure if this will fix yours as well, but the issue was mine that I wasn't running the bluetooth daemon with the experimental argument, and I assume that it limits access to some of the API's that the python code calls.

While following some of the documentation listed here, I was running into an issue where the daemon was running, but not with the --experimental tag after running 'ps aux | grep blue'. After a lot of tinkering, I found this [site](https://learn.adafruit.com/install-bluez-on-the-raspberry-pi/installation) which walks through the steps to edit the config file, and this method worked.

Hope this helps!

nhnifong, 2018-09-02:
I encountered this on a linux desktop as well where I already had bluez 5.37 running.
`systemctl status bluetooth`
shows recent log messages like
`Unable to register GATT service with handle 0x000c for device`
And a bit of googling on that error turned up the claim that 

> GATT is part of bluez experimental feature: bluetoothd -E.

The solution:
`sudo gedit /lib/systemd/system/bluetooth.service`
find the line
`ExecStart=/usr/lib/bluetooth/bluetoothd` and add ` --experimental`
reboot.

ukBaz, 2018-09-03:
Would it be worth adding some checks to the library that an expected version etc is available?
If they weren't available then give more user friendly message.

One way of getting the information in Python is:

```python
import re
import subprocess

def bluez_version():
    """
    get the version of the BlueZ daemon being used on the system
    :return: String of BlueZ version
    """
    p = subprocess.Popen(['bluetoothctl', '-v'], stdout=subprocess.PIPE)
    ver = p.communicate()
    return str(ver[0].decode().rstrip())

def bluez_experimental_mode():
    """
    Return True if the BlueZ daemon service is in experimental mode
    :return: True if experimental enabled
    """
    status = subprocess.check_output('service bluetooth status', shell=True)
    if re.search('--experimental', status.decode('utf-8')) is None:
        return False
    else:
        return True
```

closed 2020-11-02 by dhalbert