adafruit/Adafruit_Python_BluefruitLE #21: very slow response

timredfern, 2017-03-13:
I'm trying to log data from a notify characteristic that publishes a 4 byte number 15 times per second.

The library does kind of work, but:

-there is a ~10 second delay before initially receiving a notification after subscribing
-notifications arrive erratically in bunches of 2 or 3 about every second - so I guess most must be discarded

I'm using OSX sierra on a mac book pro.

I've seen similar performance in linux.

Without looking into it deeply, is this a limitation of desktop BLE libraries or in the implementation?

luopio, 2017-04-06:
I'm seeing similar performance problems also on a mac. Using a sensor that transmits 100 12bit packages per second (works just fine on Android):
```
Waiting up to 60 seconds to receive data from the device...
>> 1.51693334042 packets per second
>> 6.69153234716 packets per second
>> 6.67559440157 packets per second
>> 5.01419869372 packets per second
```

Anything that could be done to increase the throughput?

luopio, 2017-04-24:
Disregard my comment. After further trials with a node-based solution the problem was found in the firmware of the device, which dropped from 100hz to 5hz after ~1 second of operation. After fixing that I'm seeing 100Hz with both Node and Python.

gilduz, 2017-07-25:
I have a similar issue with my BLE device, but the firmware is ok. I'm able to stream in android and ios 19 byte characteristic at ~120 Hz. Using this library I'm missing really a lot of packets. I'm on MacOS. Any chance to find a solution?

closed 2020-11-02 by dhalbert