adafruit/Adafruit_Python_BluefruitLE #22: Can't reconnect to device after disconnecting from it.

bpolonia, 2017-03-21:
Is there a way to reconnect after disconnecting? I'm running into an issue where I connect to a device, start discovery, create an instance of the service, disconnect, but then if I call connect on it again it fails to connect.

HughAndersonNUS, 2017-09-24:
Hi - this is happening to me too. 
The first connect to a device works fine, I disconnect, but the second device.connect() to the same device times out. Strangely, the MAC bluetooth information pulldown says it is connected this second time.
I turned on debugging,  and the calls seem the same:
[error.txt](https://github.com/adafruit/Adafruit_Python_BluefruitLE/files/1326977/error.txt)
In the first connect, if device.is_connected is true. After the disconnect, it becomes false. On the second attempt to connect it is still false

bpolonia - did you find how to make it work properly? Any comments from anyone? - cheers Hugh

HughAndersonNUS, 2017-09-24:
Still trying to make this work, I have just tried device.connect(), device.disconnect() and then device.connect() again, and using pprint to print out the device object before and after each call:
[error1.txt](https://github.com/adafruit/Adafruit_Python_BluefruitLE/files/1327046/error1.txt)
My OS is OSX 10.12.6, and the Mac BT menu-bar reports that the connect/disconnect/connect is working OK, (i.e. the BLE device/peripheral is connected), but the code still times out.
Cheers Hugh

bpolonia, 2017-09-24:
When you disconnect the instance of the device you were connecting to seems to be lost, which is why I think you can't reconnect. The workaround I found was when I disconnected, rescanning, finding the device again and connecting again. That works.

HughAndersonNUS, 2017-09-24:
Hi Brian,
Thanks for that - after disconnecting, I now do this:
`        known_gmss = set()
        adapter.stop_scan()
        adapter.start_scan()
`
... and I can now reconnect, once the devices turn up again. Yuch :)
Anyway - thanks for the suggestion - it works for me too.
Cheers Hugh

closed 2020-11-02 by dhalbert