adafruit/Adafruit_Python_BluefruitLE #23: Incompatibility with BlueZ version >= 5.38

neomanic, 2017-07-06:
There is a known incompatible with BlueZ versions greater than 5.37, since the API was[ changed in 5.38 to fully conform to D-Bus specs](http://www.bluez.org/release-of-bluez-5-38/). 

This results in a `device.find_characteristic(uuid)` call failing because `paths = self._props.Get(_SERVICE_INTERFACE, 'Characteristics')` in `bluez_dbus/gatt.py` throws an exception with the `Characteristics` property doesn't exist.

The fix is relatively simple (although I'd ended up writing almost a complete wrapper for Bluez D-Bus by the time I figured it out), the GATT Characteristics are now objects in the tree below the Service objects, rather than being properties.

Patch: [gatt_fix_characteristics_bluez_5.38.patch.txt](https://github.com/adafruit/Adafruit_Python_BluefruitLE/files/1129534/gatt_fix_characteristics_bluez_5.38.patch.txt)

This patch could be edited to retain compatibility with pre-5.38 versions of BlueZ, it would take a version check in the setup process and alternate code paths. Happy to add those if it means the patch can be merged. In the meantime, this issue should help anyone coming fresh to this library and not wanting to revert to an older BlueZ.

qizhe, 2017-08-18:
I use bluez 5.46, everything is fine after following your instruction except when I write value into a characteristic:
        antidos_char.write_value([ord('0'), ord('1'), ord('1'), ord('i'), ord('3')])

<Adafruit_BluefruitLE.bluez_dbus.gatt.BluezGattService object at 0x7573edf0>
ERROR:dbus.connection:Unable to set arguments ([48, 49, 49, 105, 51],) according to signature 'aya{sv}': <class 'TypeError'>: More items found in D-Bus signature than in Python arguments
Traceback (most recent call last):
  File "low_level.py", line 97, in connect
    antidos_char.write_value([ord('0'), ord('1'), ord('1'), ord('i'), ord('3')])
  File "/home/pi/env/lib/python3.4/site-packages/Adafruit_BluefruitLE-0.9.9-py3.4.egg/Adafruit_BluefruitLE/bluez_dbus/gatt.py", line 85, in write_value
    self._characteristic.WriteValue(value)
  File "/usr/lib/python3/dist-packages/dbus/proxies.py", line 145, in __call__
    **keywords)
  File "/usr/lib/python3/dist-packages/dbus/connection.py", line 641, in call_blocking
    message.append(signature=signature, *args)
TypeError: More items found in D-Bus signature than in Python arguments

brimston3, 2018-10-12:
In addition to the first patch, the empty dict additional argument in BluezGattCharacteristic.read_value() and *.write_value() got it working for me. Using bluez 5.43 from debian9.

kschmelzer, 2019-08-02:
Is there any chance we can get @donatieng fix for this merged? There seems to be a known issue with the latest version of bluez that has a simple fix ready to go.

ladyada, 2019-08-02:
@kschmelzer hiya have you tested that fix?

kschmelzer, 2019-08-02:
Hi @ladyada I have tested af46b05 and can confirm that it addresses the DBUS crash with BlueZ versions >= 5.38. As @neomanic points out, this would be a breaking change and it would require the README.txt to be update to reference 5.38 as the minimum version supported. I would be happy to help if I can.

ladyada, 2019-08-02:
ok thanks, can you try making a PR with both that fix and the readme change? i will merge it :)

kschmelzer, 2019-08-02:
I will do that, thanks!

rdagger, 2019-10-06:
Any updates on the PR?

closed 2020-11-02 by dhalbert