adafruit/Adafruit_Python_BluefruitLE #28: org.freedesktop.DBus.Error.LimitsExceeded: Connection ":1.26" is not allowed to add more match rules

WALLTECH, 2017-10-10:
Thank you for opening an issue on an Adafruit Python library repository.  To
improve the speed of resolution please review the following guidelines and
common troubleshooting steps below before creating the issue:

- **Do not use GitHub issues for troubleshooting projects and issues.**  Instead use
  the forums at http://forums.adafruit.com to ask questions and troubleshoot why
  something isn't working as expected.  In many cases the problem is a common issue
  that you will more quickly receive help from the forum community.  GitHub issues
  are meant for known defects in the code.  If you don't know if there is a defect
  in the code then start with troubleshooting on the forum first.

- **If following a tutorial or guide be sure you didn't miss a step.** Carefully
  check all of the steps and commands to run have been followed.  Consult the
  forum if you're unsure or have questions about steps in a guide/tutorial.

- **For Python/Raspberry Pi projects check these very common issues to ensure they don't apply**:

  - If you are receiving an **ImportError: No module named...** error then a
    library the code depends on is not installed.  Check the tutorial/guide or
    README to ensure you have installed the necessary libraries.  Usually the
    missing library can be installed with the `pip` tool, but check the tutorial/guide
    for the exact command.  

  - **Be sure you are supplying adequate power to the board.**  Check the specs of
    your board and power in an external power supply.  In many cases just
    plugging a board into your computer is not enough to power it and other
    peripherals.

  - **Double check all soldering joints and connections.**  Flakey connections
    cause many mysterious problems.  See the [guide to excellent soldering](https://learn.adafruit.com/adafruit-guide-excellent-soldering/tools) for examples of good solder joints.

If you're sure this issue is a defect in the code and checked the steps above
please fill in the following fields to provide enough troubleshooting information.
You may delete the guideline and text above to just leave the following details:

- Platform/operating system (i.e. Raspberry Pi with Raspbian operating system,
  Windows 32-bit, Windows 64-bit, Mac OSX 64-bit, etc.):  RPI Raspbian
- Python version (run `python -version` or `python3 -version`):  3.5.3
- Error message you are receiving, including any Python exception traces:  Traceback (most recent call last):
  File "/home/pi/Adafruit_Python_BluefruitLE/examples/scan_test.py", line 56, in <module>
    ble.run_mainloop_with(main)
  File "/home/pi/Adafruit_Python_BluefruitLE/Adafruit_BluefruitLE/bluez_dbus/provider.py", line 105, in run_mainloop_with
    raise_(self._exception[1], None, self._exception[2])
  File "/home/pi/Adafruit_Python_BluefruitLE/Adafruit_BluefruitLE/bluez_dbus/provider.py", line 120, in _user_thread_main
    self._return_code = target()
  File "/home/pi/Adafruit_Python_BluefruitLE/examples/scan_test.py", line 40, in main
    for device in set(ble.find_devices()):
  File "/home/pi/Adafruit_Python_BluefruitLE/Adafruit_BluefruitLE/interfaces/provider.py", line 109, in find_devices
    devices = self.list_devices()
  File "/home/pi/Adafruit_Python_BluefruitLE/Adafruit_BluefruitLE/bluez_dbus/provider.py", line 170, in list_devices
    return map(BluezDevice, self._get_objects('org.bluez.Device1'))
  File "/home/pi/Adafruit_Python_BluefruitLE/Adafruit_BluefruitLE/bluez_dbus/device.py", line 53, in __init__
    self._props.connect_to_signal('PropertiesChanged', self._prop_changed)
  File "/usr/lib/python2.7/dist-packages/dbus/proxies.py", line 544, in connect_to_signal
    dbus_interface, **keywords)
  File "/usr/lib/python2.7/dist-packages/dbus/proxies.py", line 369, in connect_to_signal
    **keywords)
  File "/usr/lib/python2.7/dist-packages/dbus/bus.py", line 157, in add_signal_receiver
    watch = self.watch_name_owner(bus_name, callback)
  File "/usr/lib/python2.7/dist-packages/dbus/bus.py", line 373, in watch_name_owner
    return NameOwnerWatch(self, bus_name, callback)
  File "/usr/lib/python2.7/dist-packages/dbus/bus.py", line 80, in __init__
    arg0=bus_name)
  File "/usr/lib/python2.7/dist-packages/dbus/bus.py", line 160, in add_signal_receiver
    self.add_match_string(str(match))
  File "/usr/lib/python2.7/dist-packages/dbus/bus.py", line 398, in add_match_string
    BUS_DAEMON_IFACE, 'AddMatch', 's', (rule,))
  File "/usr/lib/python2.7/dist-packages/dbus/connection.py", line 651, in call_blocking
    message, timeout)
dbus.exceptions.DBusException: org.freedesktop.DBus.Error.LimitsExceeded: Connection ":1.26" is not allowed to add more match rules (increase limits in configuration file if required)
Error in atexit._run_exitfuncs:
Traceback (most recent call last):
  File "/usr/lib/python2.7/atexit.py", line 24, in _run_exitfuncs
    func(*targs, **kargs)
  File "/home/pi/Adafruit_Python_BluefruitLE/Adafruit_BluefruitLE/bluez_dbus/adapter.py", line 76, in stop_scan
    self._adapter.StopDiscovery()
  File "/usr/lib/python2.7/dist-packages/dbus/proxies.py", line 145, in __call__
    **keywords)
  File "/usr/lib/python2.7/dist-packages/dbus/connection.py", line 651, in call_blocking
    message, timeout)
DBusException: org.bluez.Error.Failed: No discovery started
Error in sys.exitfunc:
Traceback (most recent call last):
  File "/usr/lib/python2.7/atexit.py", line 24, in _run_exitfuncs
    func(*targs, **kargs)
  File "/home/pi/Adafruit_Python_BluefruitLE/Adafruit_BluefruitLE/bluez_dbus/adapter.py", line 76, in stop_scan
    self._adapter.StopDiscovery()
  File "/usr/lib/python2.7/dist-packages/dbus/proxies.py", line 145, in __call__
    **keywords)
  File "/usr/lib/python2.7/dist-packages/dbus/connection.py", line 651, in call_blocking
    message, timeout)
dbus.exceptions.DBusException: org.bluez.Error.Failed: No discovery started

- List the steps to reproduce the problem below (if possible attach code or commands
  to run):

# Search for BLE Device devices and list all that are found.
# Author: Tony DiCola
import atexit
import time

import Adafruit_BluefruitLE

interval = 5

# Get the BLE provider for the current platform.
ble = Adafruit_BluefruitLE.get_provider()

# Main function implements the program logic so it can run in a background
# thread.  Most platforms require the main thread to handle GUI events and other
# asyncronous events like BLE actions.  All of the threading logic is taken care
# of automatically though and you just need to provide a main function that uses
# the BLE provider.
def main():
    # Clear any cached data because both bluez and CoreBluetooth have issues with
    # caching data and it going stale.
    ble.clear_cached_data()

    # Get the first available BLE network adapter and make sure it's powered on.
    adapter = ble.get_default_adapter()
    adapter.power_on()

    # Start scanning with the bluetooth adapter.
    adapter.start_scan()
    # Use atexit.register to call the adapter stop_scan function before quiting.
    # This is good practice for calling cleanup code in this main function as
    # a try/finally block might not be called since this is a background thread.
    atexit.register(adapter.stop_scan)
    print('Searching for BLE devices...')
    # Enter a loop and print out whenever a new Device device is found.
    lastTime = time.time()
    while True:
        if time.time() - lastTime >= interval:
            adapter.stop_scan()
            for device in set(ble.find_devices()):
                print('Found Device: {0}'.format(device.id))
        
            print('new check')
            ble.clear_cached_data()
            lastTime = time.time()
            adapter.start_scan()
            

# Initialize the BLE system.  MUST be called before other BLE calls!
ble.initialize()

# Start the mainloop to process BLE events, and run the provided function in
# a background thread.  When the provided main function stops running, returns
# an integer status code, or throws an error the program will exit.
ble.run_mainloop_with(main)

oliviermartin, 2019-05-15:
I have just pushed a pull-request https://github.com/adafruit/Adafruit_Python_BluefruitLE/pull/43 that fixes the same issue as yours for me.

FYI, I maintain a fork of Adafruit_Python_BluefruitLE here: https://github.com/labapart/Adafruit_Python_BluefruitLE

ladyada, 2019-05-15:
@oliviermartin hiya, we don't have someone who is caring for this repo at this time - would you like to help merge in PRs here to keep this mainline going? as you can tell this code needs a little sprucing up once in a while but otherwise mostly works :)

stonehippo, 2019-07-08:
@ladyada are you still looking for someone to manage this code? I’m doing some work with RPi BLE programming right now, and I might be able to lend a hand here. I found that some of the previous PRs were required to get things working on the latest Raspbian Buster and things still aren’t 100%, maybe.

ladyada, 2019-07-08:
we'd really love any assistance you can provide, this software needs a little spit-n-polish but the bones are good :)

stonehippo, 2019-07-08:
> we'd really love any assistance you can provide, this software needs a little spit-n-polish but the bones are good :)

Great! I’ll start with a fork, bring in the open PRs, then take a look at other improvents.

ladyada, 2019-07-08:
basically just getting it working (again) would be awesome :D

oliviermartin, 2019-07-08:
I will send my changes as PR too. My [fork](https://github.com/labapart/Adafruit_Python_BluefruitLE) is relatively stable.

ladyada, 2019-07-08:
awesome, thank you

evanthomas, 2019-12-03:
@oliviermartin 's PR is good in that creates a close method for device objects that helps to stop these resources leaking. But it is only a partial fix. In the loop above, multiple objects for the same device are lost when added to a set. Yet, these objects hold resources and need to be closed by the caller. I think this is a poor design. The user should be required to call something in the device to initialise so that they are aware the object needs to be closed.

closed 2020-11-02 by dhalbert