adafruit/Adafruit_Python_BluefruitLE #30: OC_PythonException/NSException on example script

joshbode, 2017-12-02:
Hello,

I'm trying to get `AdaFruit_Python_BluefruitLE` working under macOS 10.3.1 and I'm getting the following error when trying the examples:

```
$ python3 examples/device_info.py
Using adapter: Default Adapter
Disconnecting any connected UART devices...
Searching for UART device...
2017-12-02 14:57:07.554 Python[97493:18996787] *** Terminating app due to uncaught exception 'OC_PythonException', reason: '<class 'AssertionError'>: <memory at 0x10d04fd08>'
*** First throw call stack:
(
        0   CoreFoundation                      0x00007fff4c7d62fb __exceptionPreprocess + 171
        1   libobjc.A.dylib                     0x00007fff73143c76 objc_exception_throw + 48
        2   CoreFoundation                      0x00007fff4c7d6249 -[NSException raise] + 9
        3   _objc.cpython-36m-darwin.so         0x000000010b89779e PyObjCErr_ToObjCWithGILState + 46
        4   _objc.cpython-36m-darwin.so         0x000000010b8ce832 -[OC_PythonObject forwardInvocation:] + 898
        5   CoreFoundation                      0x00007fff4c74c3ac ___forwarding___ + 748
        6   CoreFoundation                      0x00007fff4c74c038 _CF_forwarding_prep_0 + 120
        7   CoreBluetooth                       0x00007fff4c28894c -[CBCentralManager handlePeripheralDiscovered:] + 1163
        8   CoreBluetooth                       0x00007fff4c289624 -[CBCentralManager handleMsg:args:] + 543
        9   CoreBluetooth                       0x00007fff4c2978ae __30-[CBXpcConnection _handleMsg:]_block_invoke + 53
        10  libdispatch.dylib                   0x00007fff73d006cd _dispatch_call_block_and_release + 12
        11  libdispatch.dylib                   0x00007fff73cf8e88 _dispatch_client_callout + 8
        12  libdispatch.dylib                   0x00007fff73d0d356 _dispatch_queue_serial_drain + 635
        13  libdispatch.dylib                   0x00007fff73d00239 _dispatch_queue_invoke + 373
        14  libdispatch.dylib                   0x00007fff73d042fd _dispatch_main_queue_callback_4CF + 776
        15  CoreFoundation                      0x00007fff4c78e9c9 __CFRUNLOOP_IS_SERVICING_THE_MAIN_DISPATCH_QUEUE__ + 9
        16  CoreFoundation                      0x00007fff4c750c4a __CFRunLoopRun + 2586
        17  CoreFoundation                      0x00007fff4c74ffa3 CFRunLoopRunSpecific + 483
        18  Foundation                          0x00007fff4e8003f6 -[NSRunLoop(NSRunLoop) runMode:beforeDate:] + 277
        19  _objc.cpython-36m-darwin.so         0x000000010b8d6aaf ffi_call_unix64 + 79
        20  ???                                 0x00007ffee5087290 0x0 + 140732740956816
)
libc++abi.dylib: terminating with uncaught exception of type NSException
```

It might be a `PyObjC` issues, but has anyone seen/solved this?

Cheers,
Josh

joshbode, 2017-12-03:
This is being caused by incorrect Objective-C to Python UUID conversions in Python 3. Addressed in #31

masterchef8, 2018-02-08:
Hello, 

Unfortunatel, this is still happening. I have the same error when running an example of this lib on MBP touch bar 2017 with python3.

`sing adapter: Default Adapter
Disconnecting any connected UART devices...
Searching for UART device...
2018-02-08 17:55:29.416 Python[9453:1963620] *** Terminating app due to uncaught exception 'OC_PythonException', reason: '<class 'AssertionError'>: <memory at 0x1062a9708>'
*** First throw call stack:
(
	0   CoreFoundation                      0x00007fffcef592cb __exceptionPreprocess + 171
	1   libobjc.A.dylib                     0x00007fffe3d7048d objc_exception_throw + 48
	2   CoreFoundation                      0x00007fffcef59219 -[NSException raise] + 9
	3   _objc.cpython-36m-darwin.so         0x0000000104aa195e PyObjCErr_ToObjCWithGILState + 46
	4   _objc.cpython-36m-darwin.so         0x0000000104ab9c22 -[OC_PythonObject forwardInvocation:] + 898
	5   CoreFoundation                      0x00007fffceecb54a ___forwarding___ + 538
	6   CoreFoundation                      0x00007fffceecb2a8 _CF_forwarding_prep_0 + 120
	7   CoreBluetooth                       0x00007fffcea8cf7e -[CBCentralManager handlePeripheralDiscovered:] + 923
	8   CoreBluetooth                       0x00007fffcea8d6d4 -[CBCentralManager xpcConnection:didReceiveMsg:args:] + 231
	9   CoreBluetooth                       0x00007fffcea97996 __34-[CBXpcConnection handleMsg:args:]_block_invoke + 77
	10  libdispatch.dylib                   0x00007fffe4629524 _dispatch_call_block_and_release + 12
	11  libdispatch.dylib                   0x00007fffe46208fc _dispatch_client_callout + 8
	12  libdispatch.dylib                   0x00007fffe462daac _dispatch_main_queue_callback_4CF + 925
	13  CoreFoundation                      0x00007fffcef0ebc9 __CFRUNLOOP_IS_SERVICING_THE_MAIN_DISPATCH_QUEUE__ + 9
	14  CoreFoundation                      0x00007fffceecfc0d __CFRunLoopRun + 2205
	15  CoreFoundation                      0x00007fffceecf114 CFRunLoopRunSpecific + 420
	16  Foundation                          0x00007fffd08e2252 -[NSRunLoop(NSRunLoop) runMode:beforeDate:] + 277
	17  _objc.cpython-36m-darwin.so         0x0000000104ac87af ffi_call_unix64 + 79
	18  ???                                 0x00007fff5bab52a0 0x0 + 140734731342496
)
libc++abi.dylib: terminating with uncaught exception of type NSException`

joshbode, 2018-02-08:
Thanks for confirming #31 works for you, @masterchef8 :)

joshbode, 2018-06-10:
Closed via #31

closed 2018-06-10 by joshbode

arsheen, 2018-07-31:
I still can't seem to make it work.

$ sudo python3 list_uarts.py
Using adapter: Default Adapter
Searching for UART devices...
Press Ctrl-C to quit (will take ~30 seconds on OSX).
2018-07-30 18:00:42.575 python[957:61243] *** Terminating app due to uncaught exception 'OC_PythonException', reason: '<class 'AssertionError'>: <memory at 0x11200b048>'
*** First throw call stack:
(
	0   CoreFoundation                      0x00007fff4408a2db __exceptionPreprocess + 171
	1   libobjc.A.dylib                     0x00007fff6b22bc76 objc_exception_throw + 48
	2   CoreFoundation                      0x00007fff4408a229 -[NSException raise] + 9
	3   _objc.cpython-36m-darwin.so         0x0000000110bb01ce PyObjCErr_ToObjCWithGILState + 46
	4   _objc.cpython-36m-darwin.so         0x0000000110bed604 -[OC_PythonObject forwardInvocation:] + 820
	5   CoreFoundation                      0x00007fff4400055c ___forwarding___ + 748
	6   CoreFoundation                      0x00007fff440001e8 _CF_forwarding_prep_0 + 120
	7   CoreBluetooth                       0x00007fff43b1f94c -[CBCentralManager handlePeripheralDiscovered:] + 1163
	8   CoreBluetooth                       0x00007fff43b20624 -[CBCentralManager handleMsg:args:] + 543
	9   CoreBluetooth                       0x00007fff43b2e8ae __30-[CBXpcConnection _handleMsg:]_block_invoke + 53
	10  libdispatch.dylib                   0x00007fff6be135fa _dispatch_call_block_and_release + 12
	11  libdispatch.dylib                   0x00007fff6be0bdb8 _dispatch_client_callout + 8
	12  libdispatch.dylib                   0x00007fff6be20217 _dispatch_queue_serial_drain + 635
	13  libdispatch.dylib                   0x00007fff6be13166 _dispatch_queue_invoke + 373
	14  libdispatch.dylib                   0x00007fff6be17221 _dispatch_main_queue_callback_4CF + 776
	15  CoreFoundation                      0x00007fff44042c19 __CFRUNLOOP_IS_SERVICING_THE_MAIN_DISPATCH_QUEUE__ + 9
	16  CoreFoundation                      0x00007fff44004dfa __CFRunLoopRun + 2586
	17  CoreFoundation                      0x00007fff44004153 CFRunLoopRunSpecific + 483
	18  Foundation                          0x00007fff46100f26 -[NSRunLoop(NSRunLoop) runMode:beforeDate:] + 277
	19  _objc.cpython-36m-darwin.so         0x0000000110bf08c7 ffi_call_unix64 + 79
	20  ???                                 0x00007ffedfeee540 0x0 + 140732655396160
)
libc++abi.dylib: terminating with uncaught exception of type NSException
Abort trap: 6

bartmika, 2019-07-13:
Hello, I seem to have the same issue.

python3 uart_service.py 
Using adapter: Default Adapter
Disconnecting any connected UART devices...
Searching for UART device...
2019-07-13 15:49:31.887 python3[1452:59184] *** Terminating app due to uncaught exception 'OC_PythonException', reason: '<class 'AssertionError'>: <memory at 0x110792588>'
*** First throw call stack:
(
	0   CoreFoundation                      0x00000001000f6439 __exceptionPreprocess + 256
	1   libobjc.A.dylib                     0x0000000100a6a3c6 objc_exception_throw + 48
	2   CoreFoundation                      0x000000010011028d -[NSException raise] + 9
	3   _objc.cpython-36m-darwin.so         0x000000010371421e PyObjCErr_ToObjCWithGILState + 46
	4   _objc.cpython-36m-darwin.so         0x000000010374d239 -[OC_PythonObject forwardInvocation:] + 825
	5   CoreFoundation                      0x0000000100097c71 ___forwarding___ + 767
	6   CoreFoundation                      0x00000001000978e8 _CF_forwarding_prep_0 + 120
	7   CoreBluetooth                       0x000000010e75e46a -[CBCentralManager handlePeripheralDiscovered:] + 1156
	8   CoreBluetooth                       0x000000010e75f4ba -[CBCentralManager handleMsg:args:] + 536
	9   CoreBluetooth                       0x000000010e75a7db __30-[CBXpcConnection _handleMsg:]_block_invoke + 53
	10  libdispatch.dylib                   0x00000001018cd5f8 _dispatch_call_block_and_release + 12
	11  libdispatch.dylib                   0x00000001018ce63d _dispatch_client_callout + 8
	12  libdispatch.dylib                   0x00000001018d48e0 _dispatch_lane_serial_drain + 602
	13  libdispatch.dylib                   0x00000001018d53c6 _dispatch_lane_invoke + 433
	14  libdispatch.dylib                   0x00000001018d954b _dispatch_main_queue_callback_4CF + 813
	15  CoreFoundation                      0x0000000100040556 __CFRUNLOOP_IS_SERVICING_THE_MAIN_DISPATCH_QUEUE__ + 9
	16  CoreFoundation                      0x000000010003fc83 __CFRunLoopRun + 2300
	17  CoreFoundation                      0x000000010003f135 CFRunLoopRunSpecific + 459
	18  Foundation                          0x0000000104a1c7df -[NSRunLoop(NSRunLoop) runMode:beforeDate:] + 280
	19  _objc.cpython-36m-darwin.so         0x000000010374e937 ffi_call_unix64 + 79
	20  ???                                 0x0000000101e51630 0x0 + 4326757936
)
libc++abi.dylib: terminating with uncaught exception of type NSException
Abort trap: 6

joshbode, 2019-07-14:
Did you install the package via cloning the repo or via `pip`?
It looks like the version in PyPI is from before the fix in #31

bartmika, 2019-07-15:
@joshbode Yup did that and I no longer get the error. Thanks!