adafruit/Adafruit_Python_BluefruitLE #32: Occurrd Error with connected USB Serial device

SNippon, 2017-12-04:
- Platform/operating system 
  Mac OSX 64-Bit
  macOS High Sierra 10.13.1

- Python version
  python3.6.0

- Error message you are receiving, including any Python exception traces:

```
2017-12-04 13:43:00.872 python[691:5388] *** Terminating app due to uncaught exception 'OC_PythonException', reason: '<class 'ValueError'>: bytes is not a 16-char string'
*** First throw call stack:
(
	0   CoreFoundation                      0x00007fff2d1d82fb __exceptionPreprocess + 171
	1   libobjc.A.dylib                     0x00007fff53eefc76 objc_exception_throw + 48
	2   CoreFoundation                      0x00007fff2d1d8249 -[NSException raise] + 9
	3   _objc.cpython-36m-darwin.so         0x000000010b2b75ae PyObjCErr_ToObjCWithGILState + 46
	4   _objc.cpython-36m-darwin.so         0x000000010b2e4692 -[OC_PythonObject forwardInvocation:] + 898
	5   CoreFoundation                      0x00007fff2d14e3ac ___forwarding___ + 748
	6   CoreFoundation                      0x00007fff2d14e038 _CF_forwarding_prep_0 + 120
	7   CoreBluetooth                       0x00007fff2cc8a94c -[CBCentralManager handlePeripheralDiscovered:] + 1163
	8   CoreBluetooth                       0x00007fff2cc8b624 -[CBCentralManager handleMsg:args:] + 543
	9   CoreBluetooth                       0x00007fff2cc998ae __30-[CBXpcConnection _handleMsg:]_block_invoke + 53
	10  libdispatch.dylib                   0x00007fff54aac6cd _dispatch_call_block_and_release + 12
	11  libdispatch.dylib                   0x00007fff54aa4e88 _dispatch_client_callout + 8
	12  libdispatch.dylib                   0x00007fff54ab9356 _dispatch_queue_serial_drain + 635
	13  libdispatch.dylib                   0x00007fff54aac239 _dispatch_queue_invoke + 373
	14  libdispatch.dylib                   0x00007fff54ab02fd _dispatch_main_queue_callback_4CF + 776
	15  CoreFoundation                      0x00007fff2d1909c9 __CFRUNLOOP_IS_SERVICING_THE_MAIN_DISPATCH_QUEUE__ + 9
	16  CoreFoundation                      0x00007fff2d152c4a __CFRunLoopRun + 2586
	17  CoreFoundation                      0x00007fff2d151fa3 CFRunLoopRunSpecific + 483
	18  Foundation                          0x00007fff2f2023f6 -[NSRunLoop(NSRunLoop) runMode:beforeDate:] + 277
	19  _objc.cpython-36m-darwin.so         0x000000010b2f3c1f ffi_call_unix64 + 79
	20  ???                                 0x00007ffee5810400 0x0 + 140732748858368
)
libc++abi.dylib: terminating with uncaught exception of type NSException
Abort trap: 6

```

- List the steps to reproduce the problem below (if possible attach code or commands
  to run):
 It error was occurred in all of example program in this library under connected Arduino via USB.
 Its'not occured with USB memory.
 If you can fix it,I'll be glad.
 Thank you for your develop this library.

SNippon, 2017-12-04:
Sorry,This problem is not relevant with USB serial device(arduino).
This problem occured when I turn on BLE peripheral device that is zs040 with cc2541 bluetooth chip.

FNCxPro, 2018-08-03:
I am getting this error on macOS 10.13.6 as well on python 3.7

vikg, 2019-05-01:
had this error, tried running on python3 and made sure examples were in the same directory as the AdafruitBLE folder and it worked

tomasero, 2020-01-21:
Is there a way to run it using python3 without having the file inside the AdafruitBLE folder?

closed 2020-11-02 by dhalbert