adafruit/Adafruit_Python_BluefruitLE #48: Issues in macOS 10.15 Catalina

wodz94, 2019-11-15:
Hello there,
Thanks for the library! 
Unfortunately, the examples don't seem to run on macOS 10.15
Python: 3.6

All the examples throw the same class Assertion error:

** 2019-11-15 23:53:03.304 Python[798:9684] *** Terminating app due to uncaught exception 'OC_PythonException', reason: '<class 'AssertionError'>: <memory at 0x115582348>'
*** First throw call stack:
(
	0   CoreFoundation                      0x00000001000eed63 __exceptionPreprocess + 250
	1   libobjc.A.dylib                     0x0000000100ac8bd4 objc_exception_throw + 48
	2   CoreFoundation                      0x000000010010930c -[NSException raise] + 9
	3   _objc.cpython-36m-darwin.so         0x000000010269ea5e PyObjCErr_ToObjCWithGILState + 46
	4   _objc.cpython-36m-darwin.so         0x00000001026649c0 -[OC_PythonObject forwardInvocation:] + 736
	5   CoreFoundation                      0x00000001000952c5 ___forwarding___ + 829
	6   CoreFoundation                      0x0000000100094ef8 _CF_forwarding_prep_0 + 120
	7   CoreBluetooth                       0x000000010ebbf31b -[CBCentralManager handlePeripheralDiscovered:] + 1577
	8   CoreBluetooth                       0x000000010ebc0967 -[CBCentralManager handleMsg:args:] + 641
	9   CoreBluetooth                       0x000000010ebb866a -[CBManager xpcConnectionDidReceiveMsg:args:] + 177
	10  CoreBluetooth                       0x000000010ebb85ac __30-[CBXpcConnection _handleMsg:]_block_invoke + 50
	11  libdispatch.dylib                   0x0000000101070553 _dispatch_call_block_and_release + 12
	12  libdispatch.dylib                   0x00000001010714de _dispatch_client_callout + 8
	13  libdispatch.dylib                   0x0000000101076a9e _dispatch_lane_serial_drain + 597
	14  libdispatch.dylib                   0x0000000101077455 _dispatch_lane_invoke + 414
	15  libdispatch.dylib                   0x000000010107ca91 _dispatch_main_queue_callback_4CF + 783
	16  CoreFoundation                      0x000000010003e290 __CFRUNLOOP_IS_SERVICING_THE_MAIN_DISPATCH_QUEUE__ + 9
	17  CoreFoundation                      0x000000010003d9da __CFRunLoopRun + 2370
	18  CoreFoundation                      0x000000010003ce13 CFRunLoopRunSpecific + 499
	19  Foundation                          0x000000010381bcad -[NSRunLoop(NSRunLoop) runMode:beforeDate:] + 212
	20  _objc.cpython-36m-darwin.so         0x000000010269d787 ffi_call_unix64 + 79
	21  ???                                 0x00000001127984f8 0x0 + 4604921080
)
libc++abi.dylib: terminating with uncaught exception of type NSException
Abort trap: 6 **

I'd appreciate any support, Cheers!

kthakore, 2020-03-11:
I am having the same issues

wodz94, 2020-05-27:
Turns out, it only runs in python 2.7. Give it a go over there.

closed 2020-11-02 by dhalbert